LuckyX182/Filament_sensor_simplified #2: Isn't working for me

s33rw4h, 2020-04-28:
Hi,

i have a simple rolle switch configuration. Connected to pin 4 on the board and to the ground. But it doesn't seem to work. Am I doing something wrong here? Here is the log from octopi:

2020-04-28 11:03:48,644 - octoprint.printer.standard.job - INFO - Print job selected - origin: sdcard, path: ce3_bl~1.gco, owner: Admin, user: Admin
2020-04-28 11:03:51,483 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting print from SD"
2020-04-28 11:03:51,491 - octoprint.printer.standard.job - INFO - Print job started - origin: sdcard, path: ce3_bl~1.gco, owner: Admin, user: Admin
2020-04-28 11:03:51,513 - octoprint.plugins.filament - INFO - Printing started. Filament sensor enabled.
2020-04-28 11:03:51,518 - octoprint.plugin - ERROR - Error while calling plugin filamentsensorsimplified
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 138, in on_event
    elif event is Events.PRINT_STARTED and self.no_filament():
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 117, in no_filament
    return GPIO.input(self.pin) != self.switch
RuntimeError: You must setup() the GPIO channel first
2020-04-28 11:03:51,695 - octoprint.util.comm - INFO - Changing monitoring state from "Starting print from SD" to "Printing from SD"
2020-04-28 11:03:53,770 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2020-04-28 11:04:02,410 - octoprint.util.comm - INFO - Changing monitoring state from "Printing from SD" to "Cancelling"2020-04-28 11:04:02,422 - octoprint.util.comm - INFO - Force-sending M108 to the printer
2020-04-28 11:04:02,443 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: sdcard, path: ce3_bl~1.gco, owner: Admin, user: Admin
2020-04-28 11:04:02,457 - octoprint.plugins.filament - INFO - Printing stopped. Filament sensor disabled.
2020-04-28 11:04:02,460 - octoprint.plugins.filamentsensorsimplified - INFO - PrintCancelled: Disabling filament sensor.2020-04-28 11:04:02,469 - octoprint.plugin - ERROR - Error while calling plugin filamentsensorsimplified
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 173, in on_event
    GPIO.remove_event_detect(self.pin)
RuntimeError: No access to /dev/mem.  Try running as root!
2020-04-28 11:04:02,498 - octoprint.plugins.filament - INFO - Printing stopped. Filament sensor disabled.
2020-04-28 11:04:02,507 - octoprint.plugins.filamentsensorsimplified - INFO - PrintFailed: Disabling filament sensor.
2020-04-28 11:04:02,510 - octoprint.plugin - ERROR - Error while calling plugin filamentsensorsimplified
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 173, in on_event
    GPIO.remove_event_detect(self.pin)
RuntimeError: No access to /dev/mem.  Try running as root!
2020-04-28 11:06:37,985 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-04-28 11:07:37,360 - octoprint.util.comm - INFO - Telling the printer to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2020-04-28 11:07:41,303 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2823, in _readline
    ret = self._serial.readline()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4968, in readline
    c = self.read(1)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2020-04-28 11:07:41,315 - octoprint.util.comm - ERROR - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
2020-04-28 11:07:41,332 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2823)"

LuckyX182, 2020-04-28:
Hello, @s33rw4h, thank you for your issue, Firstly, thank you for providing log files. I am seeing two potential problems here 
- you are likely using the Octoprint-Filament plugin by MoonshineSG which may interfere with this plugin, simply go to Plugin Manager and turn this off and restart OctoPrint
- there is a log when Filament Sensor Simplified plugin starts, I don't see that here, may be higher in you logs, search for "Filament Sensor Simplified started" and "Filament Sensor active on GPIO Pin [YOUR_PIN]", if you can't find such log this plugin wasn't setup correctly for some reason and as a result you're getting this error.

s33rw4h, 2020-04-28:
Thanks for the quick reply. There was some filament plugin there, but it wasn't the octoprint-Filament by Moonshine. I've deleted it and tried it again. Here is the full log file.

pi@octopi:~ $ cd .octoprint
pi@octopi:~/.octoprint $ cd logs
pi@octopi:~/.octoprint/logs $ tail -F octoprint.log
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 50, in on_after_startup
    self._setup_sensor()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 63, in _setup_sensor
    GPIO.setup(self.pin, GPIO.IN, pull_up_down=GPIO.PUD_UP)
ValueError: The channel sent is invalid on a Raspberry Pi
2020-04-28 13:36:11,351 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://octopi.local/'}
2020-04-28 13:36:12,041 - octoprint.server.util.sockjs - INFO - User Admin logged in on the socket from client 2607:f2c0:e1de:a5:ac1e:fbb7:5332:eea2
2020-04-28 13:36:34,723 - octoprint.server.preemptive_cache - INFO - ... done in 23.37s
2020-04-28 13:38:00,486 - octoprint.server.util.flask - INFO - Passively logging in user Admin from 2607:f2c0:e1de:a5:ac1e:fbb7:5332:eea2
2020-04-28 13:38:00,491 - octoprint.access.users - INFO - Logged in user: Admin
2020-04-28 13:38:00,764 - octoprint.server.util.sockjs - INFO - Client connection closed: 2607:f2c0:e1de:a5:ac1e:fbb7:5332:eea2
2020-04-28 13:38:01,896 - octoprint.server.util.sockjs - INFO - New connection from client: 2607:f2c0:e1de:a5:ac1e:fbb7:5332:eea2
2020-04-28 13:38:02,061 - octoprint.server.util.flask - INFO - Passively logging in user Admin from 2607:f2c0:e1de:a5:ac1e:fbb7:5332:eea2
2020-04-28 13:38:02,063 - octoprint.access.users - INFO - Logged in user: Admin
2020-04-28 13:38:04,864 - octoprint.server.util.sockjs - INFO - User Admin logged in on the socket from client 2607:f2c0:e1de:a5:ac1e:fbb7:5332:eea2
2020-04-28 13:38:24,723 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-04-28 13:38:24,991 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-04-28 13:38:25,010 - octoprint.plugins.filamentsensorsimplified - INFO - Error: Disabling filament sensor.
2020-04-28 13:38:25,012 - octoprint.plugin - ERROR - Error while calling plugin filamentsensorsimplified
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 173, in on_event
    GPIO.remove_event_detect(self.pin)
ValueError: The channel sent is invalid on a Raspberry Pi
2020-04-28 13:38:52,335 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-04-28 13:38:52,458 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Opening serial port"
2020-04-28 13:38:52,472 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2020-04-28 13:38:52,488 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-28 13:38:52,725 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-28 13:38:57,104 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-28 13:38:57,112 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2020-04-28 13:38:57,152 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-28 13:38:57,202 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin Creality 3D (Creality3D)"
2020-04-28 13:38:57,217 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2020-04-28 13:39:30,245 - octoprint.printer.standard.job - INFO - Print job selected - origin: sdcard, path: ce3_bl~1.gco, owner: Admin, user: Admin
2020-04-28 13:41:59,096 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting print from SD"
2020-04-28 13:41:59,119 - octoprint.printer.standard.job - INFO - Print job started - origin: sdcard, path: ce3_bl~1.gco, owner: Admin, user: Admin
2020-04-28 13:41:59,142 - octoprint.plugin - ERROR - Error while calling plugin filamentsensorsimplified
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 138, in on_event
    elif event is Events.PRINT_STARTED and self.no_filament():
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 117, in no_filament
    return GPIO.input(self.pin) != self.switch
ValueError: The channel sent is invalid on a Raspberry Pi
[... cut ...]
2020-04-28 13:42:35,720 - octoprint.plugin - ERROR - Error while calling plugin filamentsensorsimplified
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 173, in on_event
    GPIO.remove_event_detect(self.pin)
ValueError: The channel sent is invalid on a Raspberry Pi

LuckyX182, 2020-04-28:
Thanks, now the problem is that pin number 4 is actually 5V, if you meant GPIO4, change it to 7 and it should work.

s33rw4h, 2020-04-28:
Thanks again for the reply, that seems to have fixed that problem. The print does not pause however when the sensor is triggered? Is it only suppose to pop up a message? (Which ends up covering the screen).

LuckyX182, 2020-04-28:
If you use octoprint only to start printing from SD card this won't work.

s33rw4h, 2020-04-28:
I took the SD card out and tried printing again, only from octoprint. it still did not pause the print. I'm using an Ender 3 that I believe supports M600. Thoughts?

LuckyX182, 2020-04-28:
Sorry that you are still having trouble. One thing you can try is issue M600 X0 Y0 command to octoprint terminal when printing. If it does park and begin filament change, could you please provide logs?

s33rw4h, 2020-04-28:
Ok I will post that shortly. Another problem is now is just triggering the sensor all the time. Here is the log:

2020-04-28 15:18:27,271 - octoprint.access.users - INFO - Logged in user: Admin
2020-04-28 15:18:30,035 - octoprint.server.util.sockjs - INFO - User Admin logged in on the socket from client 2607:f2c0:e1de:a5:b109:78fd:89b:7c81
2020-04-28 15:18:33,898 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-04-28 15:18:34,192 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Opening serial port"
2020-04-28 15:18:34,197 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2020-04-28 15:18:34,215 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-28 15:18:34,475 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-28 15:18:43,007 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-28 15:18:43,018 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2020-04-28 15:18:43,053 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-28 15:18:43,102 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin Creality 3D (Creality3D)"
2020-04-28 15:18:43,114 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2020-04-28 15:19:14,235 - octoprint.plugins.filamentsensorsimplified - INFO - Setting up sensor.
2020-04-28 15:19:14,238 - octoprint.plugins.filamentsensorsimplified - INFO - Using Board Mode
2020-04-28 15:19:14,244 - octoprint.plugins.filamentsensorsimplified - INFO - Filament Sensor active on GPIO Pin [7]
2020-04-28 15:19:46,539 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: Prints/NO_BL_TOUCH.gcode, owner: Admin, user: Admin
2020-04-28 15:19:46,570 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2020-04-28 15:19:46,590 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: Prints/NO_BL_TOUCH.gcode, owner: Admin, user: Admin
2020-04-28 15:19:46,606 - octoprint.plugins.filamentsensorsimplified - INFO - PrintStarted: Enabling filament sensor.
2020-04-28 15:19:46,624 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-04-28 15:19:46,694 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2020-04-28 15:19:48,778 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2020-04-28 15:19:49,055 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:19:49,063 - octoprint.plugins.filamentsensorsimplified - INFO - Sending out of filament GCODE
2020-04-28 15:19:50,071 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:19:51,077 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:19:52,081 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:19:53,083 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:19:54,087 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:19:55,090 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:19:56,093 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:19:57,096 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:19:58,100 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:19:59,102 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:00,106 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:01,110 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:02,113 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:03,117 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:04,121 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:05,125 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:06,129 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:07,132 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:08,134 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:09,138 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:10,141 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:11,144 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:12,146 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:13,150 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:14,155 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:15,158 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
[... cut ...]
2020-04-28 15:20:23,208 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-04-28 15:20:23,231 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: Prints/NO_BL_TOUCH.gcode, owner: Admin, user: Admin
2020-04-28 15:20:23,239 - octoprint.plugins.filamentsensorsimplified - INFO - PrintCancelled: Disabling filament sensor.
2020-04-28 15:20:23,400 - octoprint.plugins.filamentsensorsimplified - INFO - PrintFailed: Disabling filament sensor.
2020-04-28 15:20:24,213 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered

s33rw4h, 2020-04-28:
Sending M600 X0 Y0 through the terminal, or any command really, doesn't seem to be doing anything. When I click on PAUSE this is what shows up in the logs however:

2020-04-28 15:36:22,298 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2020-04-28 15:36:23,234 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2020-04-28 15:36:47,012 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: Prints/NO_BL_TOUCH.gcode, owner: Admin, user: Admin
2020-04-28 15:36:47,106 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"

LuckyX182, 2020-04-28:
That is weird, then there's some issue with your printer I'm afraid. But the sensor triggering was happening to me too, are you by any chance running sensor wiring along wires to motors? I was able to resolve this issue using shielded wire for sensor and grounding the shielding, seems interference was the cause. But your logs also showed me one bug which I'm going to fix.

s33rw4h, 2020-04-29:
I lied, i can enter commands and it does work, but only on certain things. Entering M600 does not seem to work. The only thing that seems to pause is M25 which is oddly for SD card prints. When I try to restart with M24 it gives me an error. When I try to continue the print with the octoprint button, octoprint crashes. As for the wires, they aren't running that close to the motors. I'm not sure how to shield them further?

LuckyX182, 2020-04-29:
You have to get shielded wire. I'm gonna close this as the problem does not seem to be plugin related.

closed 2020-04-29 by LuckyX182

s33rw4h, 2020-04-29:
> You have to get shielded wire. I'm gonna close this as the problem does not seem to be plugin related.

The only problem plugin related is that, at leat for the Ender3, the M600 command line does not work. The M0 command does however.

LuckyX182, 2020-04-29:
It uses M600, specifically M600 X0 Y0 as you already tried.