LuckyX182/Filament_sensor_simplified #5: Feature Request: Debounce for noisy filament sensor

mofosyne, 2020-05-12:
Was using cheap ebay filament sensor which is only just switch and am getting occational false trigger due to switch bouncing. At the very least it would be good if we can set a debounce period to avoid false trigger. This feature exist in https://github.com/kontakt/Octoprint-Filament-Reloaded however I quite like the simplicity of this plugin.

Hellrazor978, 2020-06-28:
Here is what I did to fix this. I also have a noisy switch

```
                    if self.switch is 0:
                         GPIO.add_event_detect(
                            self.pin, GPIO.RISING,
                            callback=self.sensor_callback_high,
                            bouncetime=40000
                        )
                    else:
                        GPIO.add_event_detect(
                            self.pin, GPIO.FALLING,
                            callback=self.sensor_callback_low,
                            bouncetime=40000
                        )
```
You will notice I have separated the callback functions and my bouncetime is 40 seconds. What that means is that it will not activate the callback function for another 40 seconds giving that function time to do it's thing.

My call back functions:

```
    def sensor_callback_high(self, _):
        self._logger.info("Sensor was triggered high")
        sleep(30)
        if GPIO.input(self.pin) != GPIO.HIGH:
            return self._logger.info("Sensor was false positive high")
        elif not self.changing_filament_initiated:
            self.send_out_of_filament()

    def sensor_callback_low(self, _):
        self._logger.info("Sensor was triggered low")
        sleep(30)
        if GPIO.input(self.pin) != GPIO.LOW:
            return self._logger.info("Sensor was false positive low")
        elif not self.changing_filament_initiated:
            self.send_out_of_filament()
```
What this does is first print on the log it was triggered (high or low/ground or 3V3). I did this for troubleshooting purposes.
Then it waits 30 seconds to verify the pin is indeed still in the triggered state.
If it is not, print on the log false positive.
If it is, send M600

The biggest key here is making sure your bouncetime is LONGER than your callback function.
My sensor is far enough away from the extruder I can spare 30 seconds.

dmradford, 2020-09-25:
I took a different approach to solving this solution than @Hellrazor978. I opted to repeatedly check the value of the switch to ensure the trigger was true. I also added the number of checks to the UI so it can be customized. I've submitted a PR for this solution as well.
https://github.com/LuckyX182/Filament_sensor_simplified/pull/13

LuckyX182, 2020-10-19:
The false positives problem should be fixed/improved in new version.

closed 2020-10-19 by LuckyX182