LuckyX182/Filament_sensor_simplified #9: Feature request: add an option to change the filament switch type

bvrielink, 2020-08-11:
Unfortunately this plugin doesn't work for me as my sensor reports 1 when no filament is detected and 0 if filament is present. This plugin expect it to be the other way around.  Please add an option like other filament sensor plugins to change this behavior. F.e. Filament Sensor NG calls it switch type with the options "Normally open" and "Normally closed".

Tinker0000, 2020-09-12:
You can try changing the connections on your micro switch. Usually on these micro roller switches used with runout sensors, there are three terminals on the bottom of the switch labeled: NC (NC and C terminals normally closed with the roller in its' untouched position) - NO (NC and C terminals normally open with the roller in its' untouched position) - C (common, this terminal is connected to NC with the roller up and NO with the roller pushed). The micro switch can be used in either it's NO or NC state with the roller up dependent on your choice.  If there are three wires on the switch, just swap the NC - NO connections on your board.

LuckyX182, 2020-10-19:
Option added in new version.

closed 2020-10-19 by LuckyX182