LuckyX182/Filament_sensor_simplified #10: [Feature Request] Add retraction and nozzle cooling

Error-418, 2020-09-15:
I have a tendency of falling asleep with my printer running. This causes huge issues if the filament retracts for swap but I suffer heat creep.

Any chance we can get a hot end shut-down and warm up added for extended away periods during filament run out conditions?

LuckyX182, 2020-10-03:
I don't know if this is firmware dependent but at least in my case when changing filament with M600 that's the default behaviour. I'm running Marlin.

Error-418, 2020-10-03:
I'm using Marlin as well. Unfortunately my printer is currently down due to a failed fan so I can't retest to verify this, but here's what happened (to the best of my memory).

I decided to test the run out filament by fire. Started a print and let it run for a bit to make sure it was sticking to the plate. Once it started to get a little height I called it quits and turned in for the night.

Expectation: filament runs out, OctoPrint sees this, unloads the filament in the nozzle and maintains heat to the plate until I get fresh filament loaded, and tell it to resume. (I was under the impression that Marlin had certain commands M601/603? that turns out are Průša custom commands.)

Reality: M600 is issued, and I guess has a timer baked in(?). It pulled the filament out, then tried to load the same filament back in and "completed" the print. Which turned the heat to the plate off. Woke up the next morning looking to resume printing and promptly knocked my project loose.

Sadly this discovery also revealed that my heatsink cooling fan for my hotend died so I haven't been able to print until I can be bothered to take the printer out of its cubby and install both the new fan and the buckboard for it.

Anyways, the project (as is) works as intended, just not my intention.

dmradford, 2020-10-19:
@Error-418 I believe you need to adjust your advanced pause settings in the firmware. This doesn't sound like an issue with the plugin, rather an issue with your specific firmware configuration. Can you post your Configuration_adv.h file?

LuckyX182, 2020-10-19:
Heating the plate while out of filament is potential fire hazard and I believe that's why the behaviour is  set to turn hotend and heatbed off after some time. In my case it shows heat up or something similar on screen after cool down from filament runout. How the printer reacts to M600 command is firmware controlled.

closed 2021-01-02 by LuckyX182