LuckyX182/Filament_sensor_simplified #15: Release 0.2.0 is not working have I jumped in to quickly to upgrade?

ProudAS, 2020-10-19:
I had been testing version 0.1.0 but it was reporting filament present when not and filament not present when it was. I was therefore delighted to see your update appear while I was looking at your GitHub site stating the open and close setting can be changed. I installed v 0.2.0 on OctoPrint but after setting up when the test button is clicked it remains blank, nothing reported at all in either open or closed setting.

It does state under board mode in red that "Mode has been set by third party" I don't know what the 3rd party would be.

Finally when upgrading to v 0.2.0 is still states in Plugin Manager that v 0.1.0 is installed and an update is available. I have uninstalled and reinstalled and again updated and the new options are there but it remains as v 0.1.0 in Plugin Manager stating an update to v 0.2.0 is available.

I trust this is helpful and if you believe I have simply updated to soon let me know.

dmradford, 2020-10-19:
Seconded. Test button no longer responds. Plugin no longer triggers filament out and does not execute gcode when filament is removed from sensor.

dmradford, 2020-10-20:
Confirmed that commit c18a4fb1a062bdc56664f1159df92ba4afbd8525 is the last commit that initializes in OctoPrint Version 1.4.2 running Python 2.7.13 / 3.5.3

LuckyX182, 2020-10-20:
Weird. When testing I had no issues. I will test later today.

LuckyX182, 2020-10-20:
The plugin version is definitely a bug, I forgot to update setup.py

dmradford, 2020-10-20:
@LuckyX182 You might want to roll back whatever branch is hooked up to the plugin repo until this gets sorted, or point the plugin repo to a different branch.

LuckyX182, 2020-10-20:
@dmradford luckily I managed to fix the issue, so no need to. The problem was in setting defaults. Please verify issue is fixed if you can and I will close this.

ProudAS, 2020-10-20:
I updated the app using software update as it indicated an update was available. After the update completed I reloaded OctoPrint and while the extra features were there it was still showing as v0.1.0. I then uninstalled the app restarted OctoPrint and installed again followed by a reload of OctoPrint. The app is now showing that v0.2.0 is installed. The comment in red that "Mode has been set by third party" had gone so I completed the configuration and I have just run a tested. While I have only completed the one test the app performed as expected, good work. Thanks and I trust this helps.

ProudAS, 2020-10-20:
Second test completed and again worked perfectly. Excellent app! @LuckyX182

ProudAS, 2020-10-20:
One note to make having read my posts, when I upgraded tonight and it still showed as v0.1.0 but the extra features were present as in v0.2.0 it still did not function. It was only after I removed v0.1.0 completely and clean installed v0.2.0 that resulted in every thing working. Just in case the updater side needs looked at. Cheers

LuckyX182, 2020-10-21:
@ProudAS Thanks for your quick issue creation, I wouldn't have noticed it ;) I also noticed that when searching for this plugin in Plugin Manager it still shows the latest version is 0.1.0 however I don't know how to fix it. @dmradford don't you know by any chance?

ProudAS, 2020-10-21:
When I searched for app and clean installed it showed up as v0.2.0 and reflects that one loaded. For me prior to deleting and reinstalling I had tried to upgrade from v0.1.0 to v0.2.0 and while the extra functionality appeared it still stated it was v0.1.0. It was as if the path for the upgrade is different to the clean install. I have no idea if this is right just simply reporting what I experienced in case it is helpful.

LuckyX182, 2020-10-21:
That's because I forgot to up the version number and fixed it after the release. Now it should not misbehave. Anyway, good work @ProudAS :)

dmradford, 2020-10-21:
Looks to be working correctly now. Thank you for jumping on this so quickly! The version number is showing correctly as well.

closed 2020-10-21 by LuckyX182