LuckyX182/Filament_sensor_simplified #17: Plugin does not enable pullup/pulldown on GPIO pin.

cgarry, 2020-10-26:
It took me a little while to get this working on my Pi 4B and the issue turned out to be the plugin does not appear to enable the pullup (or pulldown depending on how the 'Sensor is connected to:' field is set) for the selected GPIO pin.  In the end I had to use a GPIO that happens to have a physical pullup resistor fitted.

Am I missing something here?

Thanks,
Chris

LuckyX182, 2020-10-26:
Did you find out when using the test button or when printing?

cgarry, 2020-10-26:
This was when using the test button.  I assumed the pullup/pulldown for the GPIO pin would be enabled and then left unchanged.

LuckyX182, 2021-01-08:
Could you please provide octoprint.log file?

LazeMSS, 2021-08-31:
@cgarry : https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

LuckyX182, 2022-05-22:
Closing due to inactivity.

closed 2022-05-22 by LuckyX182