LuckyX182/Filament_sensor_simplified #18: Handle deliberate filament change during a print.

cgarry, 2020-10-26:
I did a test print where I inserted a filament change command into the GCODE when I sliced it in Cura.  This process usually works well for me.

As usual the print paused at the specified layer and I followed the prompts on the display and changed the filament.  However, once I had changed the filament and hit 'Continue' it became clear that a second M600 command had been issued by the 'Filter Sensor Simplified' because it detected an out off filament condition while I was doing the planned filament change.  This resulted in the filament I had just loaded being unloaded.

Is there anyway to prevent this from happening?

Thanks,
Chris

d8ahazard, 2020-10-27:
Wow, this was just opened yesterday? That's ironic. I also came here with the intention of requesting the same thing, as I noticed the same behavior on my Ender-5 today. 

Sensor works perfectly, it'd just be nice if the plugin had an option to disable itself if it detects a separate filament change command, and then resume monitoring once it detects the filament being re-inserted.

LuckyX182, 2021-01-08:
Fixed in 0.2.3

closed 2021-01-08 by LuckyX182

cgarry, 2021-01-08:
Excellent, thanks.  I look forward to trying it out.