LuckyX182/Filament_sensor_simplified #23: Filament switch change causes crash

rmpel, 2020-12-28:
During set-up I see the "filament detected" and "triggered" correctly when using the test button,
but as soon as I start a print -- with filament detected, of course -- a message "printer ran out of filament" is shown and octoprint crashes.

during start and tests, this shows up in the log;

`2020-12-28 21:40:12,900 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentsensorsimplified/__init__.py:125: RuntimeWarning: No channels have been set up yet - nothing to clean up!  Try cleaning up at the end of your program instead!
  GPIO.cleanup()`

when starting a print, this happens;

`2020-12-28 21:41:54,954 - octoprint.plugins.filamentsensorsimplified - INFO - PrintStarted: Enabling filament sensor.
2020-12-28 21:41:54,983 - octoprint.plugins.filamentsensorsimplified - INFO - Printing aborted: no filament detected!
2020-12-28 21:41:54,992 - octoprint.util.comm - INFO - Cancelling job on behalf of plugin filamentsensorsimplified
2020-12-28 21:41:54,998 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
2020-12-28 21:41:55,021 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: 080_mouse-0.2-20-.gcode, owner: rmpel, user: None
2020-12-28 21:41:55,560 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-12-28 21:41:55,564 - octoprint.plugins.filamentsensorsimplified - INFO - Sending out of filament GCODE: M600
2020-12-28 21:41:55,826 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2020-12-28 21:41:55,874 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintCancelling
2020-12-28 21:41:55,895 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2020-12-28 21:41:55,899 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintCancelled
2020-12-28 21:41:55,902 - octoprint.plugins.filamentsensorsimplified - INFO - PrintCancelled: Disabling filament sensor.
2020-12-28 21:41:55,932 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2020-12-28 21:41:55,936 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintFailed
2020-12-28 21:41:55,937 - octoprint.plugins.filamentsensorsimplified - INFO - PrintFailed: Disabling filament sensor.
2020-12-28 21:41:56,089 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
`

octoprint is now completely frozen, RasPi needs a reboot and the printer (Stock Melzi board) needs a reset, otherwise serial connection handshake fails.

please advise...

thank you!

Remon.

LazeMSS, 2021-01-02:
Yes I have also seen it a couple of times - https://raspi.tv/2013/rpi-gpio-basics-3-how-to-exit-gpio-programs-cleanly-avoid-warnings-and-protect-your-pi

There should be no need to use that command as far as I can understand - only when ending the program

LuckyX182, 2021-01-08:
If I don't use the cleanup() method it is not possible to switch between modes in test button functionality. Causes error: - A different mode has already been set! I removed it where it is not needed so maybe try newest version? Looking at your log it is all over the place, can you rewire it to GPIO24 and 3.3V?

rmpel, 2021-01-08:
To switch back to your plugin should not be hard, and I'm willing to do that and test it out, but, rewiring to 3.3v is kind-of a pain as the Pi is in a housing that exposes only Pin 24 and ground next to it.
Is switching to 3,3v instead of ground required? or should it work fine with ground as well?

LuckyX182, 2021-01-08:
No it is not required, I was wondering if it makes any difference.

rmpel, 2021-01-09:
The new version does not change anything for me.

I tested my switch, now using python, with

```#!/usr/bin/env python

import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BCM)
INPUT_PIN = 24

GPIO.setup(INPUT_PIN, GPIO.IN, pull_up_down=GPIO.PUD_UP)

if (GPIO.input(INPUT_PIN) == GPIO.HIGH):
	print('HIGH')
elif (GPIO.input(INPUT_PIN) == GPIO.LOW):
	print('LOW')
else:
	print('undetermined')```

When filament is present, I get "LOW", when filament is out, I get "HIGH".

So, looking in the plugin code, PUD_UP is set when power=0, which is "connected to ground" in the interface, so that seems to be in order

Board Mode:GPIO/BCM
Pin: 24
Sensor connected to: ground
Switchtype: ...

```def no_filament(self):
		return (GPIO.input(self.pin) + self.power + self.triggered) % 2 is not 0```

so, in my case; with filament present, GPIO.input() is 0, self.power is 0
no-filament condition is when 0 + 0 + self.triggered is odd.
therefore, "have filament" condition should result in 0 value, therefore "triggered" variable should be 0, which is "triggered when open"

re-evaluate; "triggered when open" == self.triggered = 0
filament present: LOW -> 0 + 0 + 0 = 0, 0 % 2 = 0 -> no_filament(self) is false, ergo filament present.
NO filament present: HIGH -> 1 + 0 + 0 = 1, 1 % 2 = 1 -> no_filament(self) is true, ergo NO filament present.

The set-up is correct.

Push button "test sensor" with filament present; 
"Sensor detected filament"
Push button "test sensor" without filament present;
"Sensor triggered"

repeated 5 times, all working perfectly.

Start print;

OctoPrint did not crash, so that is one issue less, but, got popup message

"No filament detected, print cancelled"

Restarted octoprint and tried again, again, 5 times, all the same.

Disttrack, 2021-02-25:
I have the same problem, when resume from pause after sensor triggered, octprint crashed.

```
2021-02-25 08:48:27,342 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2021-02-25 08:48:27,605 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2021-02-25 08:48:27,868 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2021-02-25 08:48:28,133 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2021-02-25 08:48:28,400 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2021-02-25 08:48:28,666 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2021-02-25 08:48:28,933 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2021-02-25 08:48:29,201 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2021-02-25 08:48:29,466 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2021-02-25 08:48:29,731 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered
2021-02-25 08:48:49,959 - octoprint.util.comm - INFO - Pausing/resuming job on behalf of user disttrack
2021-02-25 08:48:49,960 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Resuming"
2021-02-25 08:48:49,991 - octoprint.printer.standard.job - INFO - Print job resumed - origin: local, path: Untitled.gcode, owner: disttrack, user: disttrack
2021-02-25 08:48:49,996 - octoprint.plugins.filamentsensorsimplified - INFO - PrintResumed: Enabling filament sensor.
2021-02-25 08:48:50,003 - octoprint.plugins.filamentsensorsimplified - INFO - PrintResumed: Disabling filament sensor.
2021-02-25 08:48:50,168 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing"
2021-02-25 08:48:22,597 - octoprint.startup - INFO - ******************************************************************************
2021-02-25 08:48:22,600 - octoprint.startup - INFO - Starting OctoPrint 1.5.3
2021-02-25 08:48:22,601 - octoprint.startup - INFO - ******************************************************************************
```

yordanov-alexandar, 2021-04-12:
Hi 
I am writing in this issue because I think my problem is related(or the same). 
After I started using Filament Sensor Simplified I started having Octoprint crashes after the print finishes. It happens most of the time but not every time. If I login through SSH and restart Octoprint from the command line everything is ok. 
I have had this problem with other filament sensor plugins and after searching I found this as the probable reason: https://github.com/kontakt/Octoprint-Filament-Reloaded/issues/54 however pretty much every other filament sensor plugin is abandoned and there is no fix in them...
Is it possible that this is the reason for this issue?

TheRobotFactory, 2021-07-29:
I'm having the same problem, all the wiring looks ok, using the test button reports filament detected or triggered as I'd expect it to. Try to start a print and it stops with a filament out error, telling it to resume crashes octoprint.

I also notice that even if filament is present I get a triggered message after a reboot, testing again will say filament found correctly.

I'm also using gpio 24 and the gnd pin next to it. 

The Pi is running octoprint and has a pi camera installed. nothing else.

LazeMSS, 2021-08-20:
@LuckyX182 is this not fixed in the latest version?

rmpel, 2021-08-20:
body, table, td, a { font-family: 'Open Sans', 'Trebuchet MS', Trebuchet, sans-serif !important; font-size: 12px; } table.footer tr td { vertical-align: bottom; white-space: nowrap; } table.footer tr td, table.footer tr td a { color: #0365dd; font-weight: 500; font-size: 18px; text-decoration: none; } table.footer tr td a a { line-height: 30px; }

Geachte lezer,

In verband met vakantie ben ik niet in de gelegenheid uw email te beantwoorden.
Voor dringende zaken verzoek ik u contact op te nemen met mijn collega's via ***@***.*** of 030-2640605.
Vanaf 6 September ben ik weer aanwezig.
Met vriendelijke groet,
Remon Pel

	
	
	

	 

	
	www.clearsite.nl | 030 264 06 05 | bekijk ons werk

TheRobotFactory, 2021-08-20:
Last update was in Jan, I reported it 22 days ago, so no, it happens in the latest release.

LazeMSS, 2021-08-20:
@TheRobotFactory and you are running a clean octoprint image?

TheRobotFactory, 2021-08-20:
I'm running a clean octopi image (actually two of them, one each on two different printers) same issue on both printers.

Other plugins installed are...

Arc-Welder
BLTouch
Cura Thumbnails
DisplayLayerProgress
FileManager
Marlin EEPROM Editor
Preheat Button
Pushover

Disabling all other plugins during testing of the filament sensor made no difference to the results.

LazeMSS, 2021-08-20:
Strange I running it the same way although on a different pin.

Does it crash on print start or when you remove filament. Try and give me and step by step guide on how to recreate and also the OctoPrint log file.

TheRobotFactory, 2021-08-20:
What pins are you using? I'll move mine and see if it makes a difference, the other people with the same problem are all using the same pins as me. 

I'll try your pins and document step by step with my pins over the weekend.

LazeMSS, 2021-08-20:
I'm using gpio/pcm mode and pin 18 and ground on the one next to it

LazeMSS, 2021-08-31:
@TheRobotFactory  any news?

TheRobotFactory, 2021-08-31:
Yes, sorry for not getting back sooner, I tested this last weekend. I took photos as I went. 
Here's my filament sensor 
[image]
Wired to my Raspberry Pi. 
[image]
My Octoprint settings
[image]

This is what happens when I test with filament
[image]
Pull the filament
[image]

Seems perfect
click the print button
[image]
then
[image]

Have to reboot the pi to get it back. 
Moved the pins to here, updated the settings and tried again
[image]
same thing happened again.

I looped around this a few times not really knowing what to do to get it working right, then gave up and wired the runout sensors to the printer mainboards directly (which included new cables with three wires not two) and built new versions of marlin with runout detection enabled. 

There was one issue doing that, the two wires were plugged into the runout sensor where +V and GND should be, not GND and S so I had to change the wiring at the sensor end to get it working when I added the +v wire. 

I don't know if that could be the problem with the setup of this, I don't see how, there was no voltage present using only the two wires, and a continuity tester reads open circuit/closed circuit when the switch is triggered.  

Maybe you can test by moving the outside wire on your filament sensor to the other side of the plug and see if it starts crashing the pi?

Whatever. I now have two printers with two working runout sensors so I'm happy and octoprint is not involved at all  so I wont be using this plugin going forward.

TheRobotFactory, 2021-09-01:
> I can't recreate it but if you could I would really like to see the octorprint log (octoprint.log)
>
> https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299 <https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299>

I would if I could but I switched over on the 22nd and Octoprint only keeps logs for a week so I can only go back to the 25th.

LazeMSS, 2021-09-01:
@TheRobotFactory  if you rewire it :)