LuckyX182/Filament_sensor_simplified #24: Plugin problem

aquahornet, 2021-02-07:
Hi,

i installed the plugin without errors, but if i set up as you can see in the screenshot, it doesnt work. 
the sensor is connected with an gnd, 3.3v and gpio 24.

octoprint.log sais:

`2021-02-07 11:35:01,050 - tornado.access - ERROR - 500 POST /api/plugin/filamentsensorsimplified (10.0.1.21) 49.94ms
2021-02-07 11:35:01,024 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin filamentsensorsimplified
Traceback (most recent call last):
  File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 125, in on_api_command
    usage = GPIO.gpio_function(selected_pin)
RuntimeError: Not running on a RPi!
`

[image]

LazeMSS, 2021-02-15:
Are you running on a rasperry pi?

aquahornet, 2021-02-15:
Yes. i got a Raspbery Pi 4B with OS Dietpi.

LazeMSS, 2021-02-15:
yes you did a custom install - so something is missing my guess is the gpio packages or the rights to the gpio pins - notice the:
GPIO.gpio_function :  RuntimeError: Not running on a RPi!

Try running:
`sudo adduser username gpio`

but maybe have a look here: https://github.com/adafruit/Adafruit_Blinka/issues/150

LuckyX182, 2021-08-30:
Closing due to inactivity.

closed 2021-08-30 by LuckyX182