LuckyX182/Filament_sensor_simplified #25: Sometimes print is not paused

bartoszwas, 2021-02-14:
I have noticed toady that sometimes print is not paused when filament runs out. Maybe retraction was the reason, I heard two clicks of my sensor and it didn't stop...

Fredlepp, 2021-02-21:
i'm testing from an hour, and the plug-in don't stop printing. a pop up appear to advertise me  "filament run out", but the print continues.

the printer don't see the M600 command?

tried on a creality ender 3 with marlin 2.0.7

bartoszwas, 2021-02-21:
I think I know what the problem is. Plugin works but print is not paused directly after filament runout. Octorpint tries to finish sent commands or is still sending commands I'm not sure.

I have changed plugin to "Filament Sensor Reloaded" and it works fine...

Fredlepp, 2021-02-21:
i didn't modify the configuration file, but i wil try to send a M600 in the terminal to look if it make something

endurance1968, 2021-03-18:
Then the issue is closed?

Fredlepp, 2021-03-18:
yes

LazeMSS, 2021-08-21:
@Fredlepp  please the issue then :)

Fredlepp, 2021-08-21:
it's not me who has create this issue, sorry

closed 2021-08-30 by LuckyX182