LuckyX182/Filament_sensor_simplified #26: Filament change not repeatable

jburke772, 2021-02-28:
Installed plugin today and ran a test. For this test I started a print with the filament through my runout switch, but not in the actual extruder. Pulled out the filament after the first layer was done and the first filament runout was correctly triggered. I did the filament change steps and the print resumed correctly. Waited about 5 minutes and caused another filament runout to occur. Plugin did not pop up the runout message on the second trigger and the printer was never sent the code to stop. I left the filament out, thinking there might be a delay, but after 2 more layers worth of printing, the plugin still did not trigger the runout. Put the filament back in and waited 5 more minutes and tried a third attempt, still did not trigger a runout.

concres, 2021-05-08:
Printer: Anet A8

concres, 2021-05-08:
I have the same issue today when I installed the plugin, it worked only on the first time that the sensor was trigged. @jburke772, were you able to solve the problem on your printer?

I noticed that, besides my sensor is working when using the Test Button, I don't see any info on the octoprint log.

Printer info:
**Printer** -- Anet A8
**Firmware** -- Marlin 1.1.9
**OctoPrint** -- 1.6.0

Sensor info:
**Board mode** -- GPIO/BCM mode
**Pin number** -- 24
**Sensor is connected to** -- Ground
**Switch type** -- Triggered when open
**Command to send** -- M600

charredchar, 2021-05-30:
I am having the same issue, tested in the same manner.

Printer info:
Printer -- CR-10 Mini
Firmware -- Klipper
OctoPrint -- 1.6.1

Sensor info:
Board mode -- Physical/Board mode (can not change this because of another plugin)
Pin number -- 15
Sensor is connected to -- Ground
Switch type -- Triggered when open
Command to send -- M600

LuckyX182, 2021-09-03:
Fixed thanks to @mike-schiller

closed 2021-09-03 by LuckyX182