LuckyX182/Filament_sensor_simplified #27: seemed to work but printer ended up printing air...

jtannenba, 2021-03-03:
After the filament change, the extruder backed the filament so when it went back to print, it was printing with air.  Please tell me the exact steps to using this.  I have new firmware and printer recognizes M600.  I like the fact when it ran out it backed it out, but after reloading (I always push the filament till it oozes) but it backed it out most of the way again.  What did I do wrong or was this a bug?  I was 27.5 hrs on a 28 hr print.

endurance1968, 2021-03-18:
Did you check https://marlinfw.org/docs/gcode/M600.html especially the optional parameters (if you are using Marlin).
For klipper check also https://github.com/KevinOConnor/klipper/issues/149

jtannenba, 2021-03-18:
Yes, and I am still confused.  Should I have not pushed the filament all the way in?

closed 2022-07-09 by LuckyX182