LuckyX182/Filament_sensor_simplified #29: alternative to m600 for ender 5 +

turbodlr, 2021-03-18:
do you have alternate command for ender5+?  the below command is ignored by it and by googling it I see it is a known issue
M600 X0 Y0

I am gcode dummy so any help you can give would be much appreciated!

turbodlr, 2021-03-18:
perhaps an M25 which maybe would then run whatever you have in your pause gcode which in my case is:
M117 Print Paused  ; Comment
G91 Z              ; Set to relative positioning mode
G1 Z15             ; Move hotend away from part by 15mm
G90 Z              ; Set to absolute positioning mode

and since I have direct drive I could then go manually load filament and prime via my knob I have on stepper.

LazeMSS, 2021-08-20:
I had the same problems with my printer - i'm working on a PR that will allow the plugin to use the native OctoPrint pause function - you can test it here: https://github.com/LazeMSS/Filament_sensor_simplified/tree/NativeOPPause

turbodlr, 2021-08-20:
That is what I am using.  What is different that you introduced?  Also did you figure out someway for octoprint to detect signal from stock runout sensor?

 

Here is how I have mine setup.

 

 

But the problem is the filament sensor on the printer is not detectable enough by the pi.  So we put an inline optical sensor in between that sensor and the printers board to pickup and amplify that signal for the pi.. works well now.

 

 

Just in case you ever need it, here is a schematic and the instructions for the circuit:

 

 

Using a PC817 optical isolator.  The optical isolator minimizes any impact  to the native signaling from the filament sensor to the controller while also isolating it from the Raspberry Pi.

 

The colors may be different for different printers but on my Creality Ender 5 Plus you need to splice in to the yellow (or signal) wire and connect it to pin 1 on the PC817 (the anode side of the LED) and splice the black (or ground) wire and connect it to a 1K resistor and then to pin 2 (the cathode side of the LED) and connect the transistor side to the Raspberry Pi as shown below.  I used wired connectors to expose the signal from the filament sensor so I did not have to actually splice the original wiring.

 

Then connect the collector (Pin 4) to pin 18 (GPIO.5 / BCM 24) on the Raspberry Pi and the emitter (pin 3) to pin 20 (ground) on the Raspberry Pi.  And then configure the filament sensor plugin using those pins.

 

 

Here are the settings I used for OctoPrint:

 

 

 

In the before print job is resumed script it really needs to know what the previous bed temp (M190 S65) and the extruder temp (M109 S220) is but I have not found a way to get those values from OctoPrint. So I didn’t have the pause cool it.. Also used a splitter to put the contraption below inline with the sensor on the printer, just unplugged the sensor, put the splitter in and plugged the sensor into the splitter.

LazeMSS, 2021-08-20:
When you reply by email all images are lost etc.

Bit yoy problems seem to be something different.

Joost79, 2022-01-25:
@turbodlr Have you got it working? Can you post your setup here without the use of images (not visible)

Thanks

turbodlr, 2022-01-25:
[filamentSensorOctoprint.pdf](https://github.com/LuckyX182/Filament_sensor_simplified/files/7937691/filamentSensorOctoprint.pdf)

Joost79, 2022-01-25:
Thanks, but that is a different sensor entirely.  Does it not work with a normal microswitch sensor like in the link below?

https://nl.aliexpress.com/item/32917462715.html?gatewayAdapt=glo2nld&spm=a2g0o.search0304.0.0.1736290aYIYGn8&algo_pvid=afc01629-4906-4d5e-8cf1-5b1b689b975c&algo_exp_id=afc01629-4906-4d5e-8cf1-5b1b689b975c-4

turbodlr, 2022-01-26:
It is the stock sensor. Just had to build that circuit which plugs in
between stock sensor and main board. Then goes to pi gpio. Let me take some
pictures tomorrow

turbodlr, 2022-01-27:
https://nl.aliexpress.com/item/32917462715.html?gatewayAdapt=glo2nld <https://nl.aliexpress.com/item/32917462715.html?gatewayAdapt=glo2nld&spm=a2g0o.search0304.0.0.1736290aYIYGn8&algo_pvid=afc01629-4906-4d5e-8cf1-5b1b689b975c&algo_exp_id=afc01629-4906-4d5e-8cf1-5b1b689b975c-4> &spm=a2g0o.search0304.0.0.1736290aYIYGn8&algo_pvid=afc01629-4906-4d5e-8cf1-5b1b689b975c&algo_exp_id=afc01629-4906-4d5e-8cf1-5b1b689b975c-4

Joost79, 2022-01-29:
Thanks, but sending pictures via email to the thread doesnt work. They simply do not show. Besides the sensor is working, but my printer isnt able to recognize the M600 (or any other Pause) commands for whatever reason.

LuckyX182, 2022-01-31:
@Joost79 Not all printers support M600 command. To enable it you have to reflash your printers firmware with M600 enabled.

turbodlr, 2022-01-31:
what about using m25

Joost79, 2022-02-01:
Well I have flashed my ender 3 pro to marlin 2.0, which supports the M600 command. But as said it doesn’t work. It only shows the notification in octoprint.

LuckyX182, 2022-02-07:
I will be releasing new version in a couple of days, stay tuned

LuckyX182, 2022-02-09:
Guys, check out 0.3.0 version, thanks

closed 2022-05-22 by LuckyX182