LuckyX182/Filament_sensor_simplified #30: Printer supports M600 but plugin will not work

isgallagher, 2021-03-28:
Bug:

When connecting the serial to the printer, I get the following notification:

> Filament sensor simplified
> M600 gcode command is not enabled on this printer! This plugin won't work.
> 

I checked through the terminal log and see the command was sent as M603.  The printer does not support M603 but does support M600 and M108.

> Send: M603
> Recv: echo:Unknown command: "M603"

Firmware version:
> Marlin TH3D U1.R2.B5 

Printer:
> Ender 3

LazeMSS, 2021-08-20:
I had the same problems with my printer - i'm working on a PR that will allow the plugin to use the native OctoPrint pause function - you can test it here: https://github.com/LazeMSS/Filament_sensor_simplified/tree/NativeOPPause

LazeMSS, 2021-08-21:
You can try installing using URL install in OctoPrint and use: https://github.com/LazeMSS/Filament_sensor_simplified/archive/refs/heads/NativeOPPause.zip

Joost79, 2022-01-25:
Have you ever got this working? I have the same issue, octoprint does not recognize the pauze command. Got the sensor to work, but it won't stop the print because of the unknown commands. Also tried the other M0, M1 commands which are "blocked commands" according to octoprint.

LuckyX182, 2022-05-22:
M603 check has been removed a while ago so this won't cause an issue now.

closed 2022-05-22 by LuckyX182