LuckyX182/Filament_sensor_simplified #33: Some instructions on what settings to use for 3-pin switch sensors

melyux, 2021-04-28:
I have a 3-pin microswitch filament sensor, the basic one you find everywhere. It's connected to 3V3, GND, and a GPIO pin. The instructions and configuration page don't clearly indicate what to choose for this: "Sensor is connected to" = ???

Ground or 3V3? It's not clear. It's connected to both. I still don't know which one to use.

melyux, 2021-04-29:
I used GND, "the safer option", and it seems to work well. Not sure if that would warrant a third choice to the list besides 3V3 and GND ("both"?) to disable the pull-ups when both a reference 3V3 and GND are provided.

InnoProd, 2021-08-18:
Thank you. Being new to RPi and Octo, I had the same question. So to be clear, you are using the microswitch as an SPST switch to simply connect GND to the GPIO pin?

LazeMSS, 2021-08-31:
Its not part of this plugin - but there are several guides - https://www.circuitbasics.com/how-to-set-up-buttons-and-switches-on-the-raspberry-pi/

closed 2021-08-31 by LazeMSS

InnoProd, 2021-08-31:
Thank you for the pointers! I found that using a N.O. switch resulted in a more stable setup. A N.C. curcuit allowed phantom triggers through my unsheilded cabling, while a N.O. configuration closes the circuit with filament present creating a more stable signal pattern.

melyux, 2021-08-31:
The issue was closed, but was anything added to the code or documentation to solve the issue of what to do in the case I'd described?

LazeMSS, 2021-08-31:
It was closed because it's not a software problem but a hardware issue. It's pretty hard to support all different hardware setup and combinations.

Did the link not help you or have you figured it out?

melyux, 2021-08-31:
I used GND as the selection and it worked. But it was a complete crapshoot guess and I got lucky. The software and documentation implies that you connect the switch between only 2 pins: GPIO + (one of GND or 3V3). There is no clear option for when you do the more electrically robust thing, which is to use 3 pins: GPIO, 3V3, _and_ GND, thus overriding the internal pull-ups.

LazeMSS, 2021-08-31:
Will try and write a better guide soon