LuckyX182/Filament_sensor_simplified #35: Please make 5-times recheck of sensor configurable or add a minimum timeframe

NilsRo, 2021-07-04:
Hi,

I am using an Anycubic Mega S sensor connected to my Raspi, also added shielding and do not have motorwires around. Actually in very rare cases the sensor accidental triggers. I think during 200h printing I have one false positive. But to have no one it would be nice to use the time remaining using the filament between extruder and the filament sensor like with the following additional configuration for the plugin:

1. Amount of seconds the sensor must be open
2. Amount of checks that have to be opened (more than 5 if the user like to have more checks)

This would result into a more stable detection if for any reason the sensor does not work like expected.

Cheers,
Nils

LuckyX182, 2021-08-30:
I would not like to add additional config just to make the plugin work, we have to find out the root issue why we get false positives.

LuckyX182, 2022-01-31:
Additional 5 time check was added while back.

closed 2022-01-31 by LuckyX182