LuckyX182/Filament_sensor_simplified #36: Performing filament change directly on printer screen retriggers filament change exactly afterwards.

melvinmicallef, 2021-07-13:
My issue is that when my printer is printing and I perform a filament change through the printer's screen (not sending M600 from octoprint), when the filament is changed and loaded, this plugin sends M600 again. Does the printer notify octoprint that a filament change was started from the printer? If so, how easy is it to fix this issue please?

LuckyX182, 2021-09-03:
I don't think that Octoprint knows when changing filament manually - the printer just stops accepting gcode commands and starts accepting more after the M600 is finished but I will investigate more if there's something to be done in this case.

melvinmicallef, 2021-09-10:
understood :) go ahead and close this issue if you do not see an easy solution. Thank you

closed 2022-02-12 by LuckyX182