LuckyX182/Filament_sensor_simplified #38: Filament Sensor simplified crashes Octoprint when I cancel a print

Koubiacz, 2021-08-22:
Hi,

I have a problem with the plugin.
When I cancel a print, octoprint doesn't respond anymore.

On the Raspberry,  when I do a "systemctl status octoprint", Linux tells me that the service is stopped (exited)

If I deactivate the plugin I don't have the problem.

logs: 

[log-octoprint-before-crash.txt](https://github.com/LuckyX182/Filament_sensor_simplified/files/7027314/log-octoprint-before-crash.txt)

[serial.log](https://github.com/LuckyX182/Filament_sensor_simplified/files/7027291/serial.log)

I didn't find anything else in the system logs

thanks

LazeMSS, 2021-08-31:
Please provide the logfiles from OctoPrint as descried here: https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

LuckyX182, 2021-09-03:
I couldn't replicate your issue on my test PI, does this happen only when cancelling print?

Koubiacz, 2021-09-08:
Hi,

Yes, the problem occurs only when I cancel a print after 5 minutes or more.