LuckyX182/Filament_sensor_simplified #43: Plugin Interferes with mid print filament change

LewisSpring, 2021-12-24:
Hello,
Would it be possible to have the script automatically temporarily disable itself during a M600 in Gcode for multi-colour prints?
I've found that as I'm removing the filament from the printer during the GCode-triggered M600, It seems to activate another M600.
It didn't seem to cause the filament to retract like the first time M600 was ran, but everything else seemed to have happened multiple times.

[image]
These happened during the print, I think.

Thank you?

LuckyX182, 2022-02-12:
Hello, please try with the latest version.

Luke

LewisSpring, 2022-02-13:
Hi there Luke,
I haven't printed for a while, but I'll get back to you as soon as possible.

LewisSpring, 2022-03-03:
Hi Luke, 
I did a print with the new version:
Seems to work great! No interference as far as I can tell.

Cheers!

LuckyX182, 2022-03-09:
Superb!

closed 2022-03-09 by LuckyX182