LuckyX182/Filament_sensor_simplified #44: Not working….

heizie, 2022-01-05:
I’ve tested the sensor. The plugin shows correct state of the filament. And the icon on top of the octoprint’s ui shows black when filament available and marked as red when not. And i’ve tried sending m600 to print, it react as expected. Everything while testing seems right.

BUT! When i pull out the testing filament while printing, the print won’t pause. Neither in m600 mode nor octoprint pause….

What’s wrong?

Joost79, 2022-01-30:
I have the same issue (running marlin 2.0), octoprint/marlin doesnt respond to M600 or the M25 command. How do i enable the advanced pause feature in marlin?

Joost79, 2022-02-13:
Just updated the mod and tried a print when i knew it would run out of filament. It nicely shows the message when it has run out, but there seems to be no response to the M600 X0 Y0 command. I will try the other commands as well, but this is not working on octoprint with an ender 3 pro on marlin 2.0 (which is able to handle the command).

LuckyX182, 2022-02-13:
@Joost79 does the printer initiate filament change when you issue M600 manually through terminal?

LuckyX182, 2022-07-09:
Closing due to inactivity.

closed 2022-07-09 by LuckyX182