LuckyX182/Filament_sensor_simplified #45: Continues messages: "Filament sensor simplified Printer ran out of filament!"

Joost79, 2022-01-24:
Hi there, just installed the filament sensor simplified. But it keeps popping up notifications (that the filament ran out) however, the printer is not being paused (using the M600 command and have the marlin 2.0 firmware). Also see below screenshot.

Is this what you meant by the interference and to use shielded cable? And how come the printer is not responding to the the message/actually being paused?

[image]

Joost79, 2022-01-24:
In addition, the icon actually shows the correct status, if there is filament the pop ups are coming every minute or so. The printer is not pausing though. But the icon is showing as it should. When there is no more filament, the status messages are still there, the icon now shows there is no filament (which is now correct). Jet the printer is not pausing or doing anything...

LuckyX182, 2022-02-12:
Hello, please try with the latest version.

Luke

LuckyX182, 2022-07-09:
Closing due to inactivity.

closed 2022-07-09 by LuckyX182