LuckyX182/Filament_sensor_simplified #47: Plugin breaks Enclosure Plugin GPIO detection

NilsRo, 2022-02-12:
I do not know which Plugin caused the issue with Enclosure Plugin but if I disabled Filament Sensor Simplified Enclosure is working again. Also the issue raised (not 100% sure) with the last update of the Filament Sensor Simplified plugin. So to solve this I disabled Filament Sensor Simplified and everything is working again. Now downgraded to version 0.2.3 and everything is fine also. Will stay on this until the issue is solved.

See here for Enclosure Error description: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/476

LuckyX182, 2022-02-12:
Hello, sorry for inconvencies, can you specify Enclosure plugin version and provide octoprint.log file?

NilsRo, 2022-02-12:
No problem.

Logs: https://github.com/vitormhenrique/OctoPrint-Enclosure/files/8044997/Enclosure.Logs.log
Enclosure Plugin (4.13.2)

LuckyX182, 2022-02-12:
I think I fixed it, please download plugin from this URL: https://github.com/LuckyX182/Filament_sensor_simplified/archive/refs/heads/develop.zip
and let me know if the problem is resolved.

Luke

NilsRo, 2022-02-12:
Yes, everything is working fine with that version.

closed 2022-02-12 by NilsRo