LuckyX182/Filament_sensor_simplified #49: after update no function

Hirnheiner2, 2022-02-17:
Hi... 
After I installed octoprint and filament sensor simplified updates, the sensor no longer works. board mode, pin 7, I use the Sentinel Filament Detector from Dyze as the sensor

LuckyX182, 2022-03-09:
Can you specify version a nd provide octoprint.log file?

Hirnheiner2, 2022-03-09:
Hi... thx for your help: Version 1.7.3

Hirnheiner2, 2022-03-09:
2022-03-08 18:50:44,433 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 18:50:44,440 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2022-03-08 18:50:44,440 - octoprint.server - INFO - OctoPrint 1.7.3
2022-03-08 18:50:44,445 - octoprint.plugin.core - INFO - 21 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Event Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/eventmanager
|  Filament Sensor Simplified (0.3.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentsensorsimplified
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.10.11) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  Fix CBD Firmware Plugin (0.4.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_fixcbdfirmware
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  Pi Support Plugin (2021.10.28) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  Pushover (0.3.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pushover
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2022-03-08 18:50:44,464 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 915718144
|  os:
|    bits: 32
|    id: linux
|    platform: linux
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Plus Rev 1.3
|      octopi_version: 0.18.0
|      throttle_state: '0x0'
|  python:
|    pip: 20.3.3
|    version: 3.7.3
|    virtualenv: /home/pi/oprint
2022-03-08 18:50:44,464 - octoprint.server - INFO - ------------------------------------------------------------------------------
2022-03-08 19:05:44,434 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 19:20:44,436 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 19:35:44,437 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 19:50:44,438 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 20:05:44,440 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 20:20:44,441 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 20:28:53,664 - octoprint.plugins.fixcbdfirmware - INFO - Replacing G28 X0 Y0 with G28 X0 Y0
2022-03-08 20:28:54,089 - octoprint.util.comm - INFO - Finished in 34059.573 s.
2022-03-08 20:28:54,090 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2022-03-08 20:28:54,098 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: 
2022-03-08 20:28:54,110 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 
2022-03-08 20:28:54,171 - octoprint.plugins.fixcbdfirmware - INFO - Replacing G28 with G28
2022-03-08 20:29:03,768 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2022-03-08 20:29:19,478 - octoprint.filemanager.analysis - INFO - Analysis of entry local:CFFFP_
2022-03-08 20:35:44,442 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 20:50:44,443 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 21:05:44,445 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 21:20:44,446 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 21:35:44,447 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 21:50:44,449 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 22:05:44,450 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 22:20:44,451 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-03-08 22:35:44,453 - octoprint.server.heartbeat - INFO - Server heartbeat <3
[... cut ...]
2022-03-09 09:42:42,007 - octoprint.server.util.sockjs - INFO - User csr logged in on the socket from client ::ffff:
2022-03-09 09:42:42,934 - octoprint.plugins.filamentsensorsimplified - INFO - Filament detected
2022-03-09 09:48:36,943 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor callback called
2022-03-09 09:48:36,946 - octoprint.plugins.filamentsensorsimplified - INFO - Reading sensor values
2022-03-09 09:48:38,951 - octoprint.plugins.filamentsensorsimplified - INFO - Filament not detected
2022-03-09 09:48:38,952 - octoprint.plugins.filamentsensorsimplified - INFO - Sensor was triggered

LuckyX182, 2022-07-09:
Can you try to reinstall the plugin?