LuckyX182/Filament_sensor_simplified #50: plugin sends G1 X0 Y0 on "octoprint pause" mode

cchaz003, 2022-02-19:
I'm not sure if this is intended/if there is a reason for this that I'm not understanding but it looks like this plugin is hard-coded to send "G1 X0 Y0" _and then_ triggers the Octoprint pause script. Unless I'm missing something, I would assume that the best course of action if Octoprint pause is selected would be to only trigger the Octoprint pause script and let the user decide what the printer should do from there. 

Thanks in advance!

LuckyX182, 2022-03-09:
Yes, it is hardcoded, I did this because it would pause on last position melting away the model and that way it is impossible to do any filament change.

cchaz003, 2022-03-09:
The issue that I faced with this (any why I suggest removing it and leaving it to the user to decide in their own on-pause script) is that when I pause for a filament change, the first thing I need to do is a retraction move and then I can move the nozzle out of the way. Without the retraction first, the nozzle moves and drags a string along with it. Ideally I retract, lift z just a tiny bit and then move to a safe 0,0 type location. This is what I've coded into my on-pause script but with your hard-coded bit running first that retract doesn't happen until it's already moved to 0,0 with a big string. 

I forked your script and removed the 0,0 command and it's working well (and thank you for the great plugin). If you have a good reason and no one else has an issue with the hard-coded bit then I have no reason to make you change it but it definitely threw me for a loop until I dug into what you were doing so maybe this discussion can help someone else in the future.

LuckyX182, 2022-03-28:
I didn't want to make it over-complicated by asking user to configure everything, but your use case makes me think that it probably should be configurable

unitrix0, 2022-04-28:
Agreed this should be removed for multiple reasons:

- If the user selects the "Octoprint pause" option it's understood that he has to care himself about the pause action in Octoprint's GCODE settings. 
- Sending GCODEs to a printer you don't know could cause further issues. You never know how much somebody has modded theyr printer. 
- Doing a simple `G1 X0 Y0` is slow and there is also no retraction which causes stringing

reddai75, 2022-07-04:
I'm struggling too with the hardcoded G1 X0 Y0, as it is necessary to retract, do a z+10 at least and then move to 0,0 ... Will you implement a custom gcode option in the future ?
thanks!

LuckyX182, 2022-07-09:
Removed hardcoded gcode in 0.3.2

closed 2022-07-09 by LuckyX182