LuckyX182/Filament_sensor_simplified #51: Microcuts detected as valid to consider a valid filament end

zurbaran, 2022-02-19:
It seems that the filament sensor sends microcuts, which causes the printer, octoprint, to detect that the filament has run out for a few thousandths of a second and pause printing. Could it be configured the times that it must be in a state change so that it considers it real? Marlin himself does not detect the end of the filament for me because he must be filtering those microcuts, but the raspberry, octoprint, if he thinks he has received the signal that the filament is over.

fraintt, 2022-03-02:
Hi, 
i see this as well, but only with the new version. It was working fine with 0.2.3, but after the Update the sensor gets triggered 3-4 in a 1 hour print altought the filament is still present..

kami83, 2022-04-07:
Same Problem here Andy solution?? BG kami

sixtysixdutch, 2022-05-02:
+1 also seeing the same issue using a basic microswitch type sensor

LuckyX182, 2022-05-22:
there must be 10 consequent reads to consider filament end as valid to prevent false positives, will try to test

LuckyX182, 2022-07-09:
Hi guys, as @CMR-DEV mentioned this is possibly fixed and now available in 0.3.2 version, please try updating