LuckyX182/Filament_sensor_simplified #52: will not detect the switch

VVildFire1, 2022-04-04:
will not detect the switch

[image]
h

so I have a VSG    V is on 17   S is on 19 and G is on 20 the LED on my switch lights up like it should be the program when testing never changes.  I'm using the old cr10 s5 filament switch and board to go into the pi.. any ideas?

LuckyX182, 2022-05-22:
you can try GPIO mode, other than that I have no suggestions what you can do

ShapeShifter42, 2023-02-13:
I had the problem where using the “Test Sensor” button during initial setup always said that the sensor was triggered, regardless of the pin/sensor/switch type settings I tried. 

I saved my initial configuration anyway, restarted the RPi, and went back to the configuration screen, at which point the “Test Sensor” button worked as expected.