LuckyX182/Filament_sensor_simplified #53: Causing OctoPrint 1.8.0 to crash (server not starting)

IHateNameStealers, 2022-05-18:
After updating to the new 1.8.0 version of OctoPrint, OctoPrint's server does not startup anymore until I disable "Filament Sensor Simplified".

Not sure if it's only with my setup.

Anyone else having the same issue?

TrueKillRob, 2022-05-19:
I have the same problem: after updating my OctoPrint from 1.7.3 to 1.8, it crashes with my Ender 7 setup. On my Ender 3 setup it runs fine.

LuckyX182, 2022-05-22:
Tested 0.3.1 version of plugin on 1.8.0 OctoPrint with no issues. What version of plugin are you running? Can you provide octoprint.log file?

der-knusprige, 2022-05-22:
having the same problem on 1.8.0, the server wont start with this plugin

log:

2022-05-22 21:02:55,544 - octoprint.startup - INFO - ******************************************************************************
2022-05-22 21:02:55,553 - octoprint.startup - INFO - Starting OctoPrint 1.8.0
2022-05-22 21:02:55,555 - octoprint.startup - INFO - ******************************************************************************
2022-05-22 21:02:59,016 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-05-22 21:03:10,072 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2022-05-22 21:03:10,072 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2022-05-22 21:03:10,073 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2022-05-22 21:03:10,101 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2022-05-22 21:03:13,654 - octoprint.startup - INFO - Added settings overlay from plugin consolidate_temp_control
2022-05-22 21:03:13,655 - octoprint.plugin.core - INFO - Found 23 plugin(s) providing 23 mixin implementations, 49 hook handlers
2022-05-22 21:03:14,069 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2022-05-22 21:03:14,092 - octoprint.server - INFO - Intermediary server started
2022-05-22 21:03:14,145 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2022-05-22 21:03:14,227 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2022-05-22 21:03:14,237 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2022-05-22 21:03:14,238 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2022-05-22 21:03:14,239 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2022-05-22 21:03:14,240 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2022-05-22 21:03:14,240 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')")
2022-05-22 21:03:14,241 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2022-05-22 21:03:14,241 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_GRANT (needs: "Need(method='role', value='plugin_appkeys_user')")
2022-05-22 21:03:14,242 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2022-05-22 21:03:14,244 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2022-05-22 21:03:14,247 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2022-05-22 21:03:14,250 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: "Need(method='role', value='plugin_pi_support_check')")
2022-05-22 21:03:14,254 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_LIST (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2022-05-22 21:03:14,256 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2022-05-22 21:03:14,260 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_install'), Need(method='role', value='plugin_pluginmanager_manage')")
2022-05-22 21:03:14,262 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2022-05-22 21:03:14,264 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2022-05-22 21:03:14,269 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2022-05-22 21:03:15,689 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python3 -m pip" as command to invoke pip
2022-05-22 21:03:16,870 - octoprint.util.pip - INFO - Version of pip is 20.3.3
[... cut ...]
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/types.py", line 1432, in get_blueprint
    f = getattr(self, member)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 68, in setting_cmd_action
    return int(self._settings.get(["cmd_action"]))
ValueError: invalid literal for int() with base 10: 'opnative'

TrueKillRob, 2022-05-22:
Hi,

I have exact the same error message?
If you need a complete new Log, I can generate one after my print is finished.
(I disabled your plug-in as a workaround)

"/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentsensorsimplified/**init**.py", line 68, in setting_cmd_action return int(self._settings.get(["cmd_action"])) ValueError: invalid literal for int() with base 10: 'opnative'

LuckyX182, 2022-05-23:
That looks weird indeed. I used opnative as option in february when developing but it was changed to 0 and 1 values very quickly and didn't make it into released code. Can you try reinstalling the plugin along with deleting plugin data?

TrueKillRob, 2022-05-23:
This was the solution!!!! Uninstall with Cleanup and reinstall :-)
I only have to reconfiugre it, but this is simple...

Only for your info. The Install-Log:

Installiere Plugin "Filament Sensor Simplified" von https://github.com/LuckyX182/Filament_sensor_simplified/archive/master.zip...
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpdl2yqgw4/Filament_sensor_simplified-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpdl2yqgw4/Filament_sensor_simplified-master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from Filament-sensor-simplified==0.3.1) (1.8.0)
Requirement already satisfied: RPi.GPIO in /home/pi/oprint/lib/python3.7/site-packages (from Filament-sensor-simplified==0.3.1) (0.7.0)
Requirement already satisfied: pathvalidate<3,>=2.4.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (2.5.0)
Requirement already satisfied: colorlog<7,>=6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (6.6.0)
Requirement already satisfied: unidecode in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (0.4.21)
Requirement already satisfied: Click<9,>=8.0.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (8.1.3)
Requirement already satisfied: Flask-Babel<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (2.0.0)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (2.8.5)
Requirement already satisfied: sarge==0.1.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (0.1.6)
Requirement already satisfied: flask<3,>=2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (2.1.2)
Requirement already satisfied: watchdog<2,>=1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (1.0.2)
Requirement already satisfied: wrapt<1.14,>=1.13.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (1.13.3)
Requirement already satisfied: tornado<7,>=6.0.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (6.1)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (0.5.0)
Requirement already satisfied: zeroconf<0.34,>=0.33 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (0.33.4)
Requirement already satisfied: netaddr<0.9,>=0.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (0.8.0)
Requirement already satisfied: feedparser<7,>=6.0.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (6.0.8)
Requirement already satisfied: websocket-client<2,>=1.2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (1.3.2)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (2021.2.23)
Requirement already satisfied: markdown<4,>=3.2.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (3.3.7)
Requirement already satisfied: regex in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (2020.11.13)
Requirement already satisfied: werkzeug<2.1,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (2.0.3)
Requirement already satisfied: cachelib<0.3,>=0.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (0.2.0)
Requirement already satisfied: netifaces<1,>=0.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (0.11.0)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (1.0.7)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (0.18.2)
Requirement already satisfied: zipstream-ng<2.0.0,>=1.3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (1.3.4)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (2021.10.11)
Requirement already satisfied: frozendict<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (2.3.2)
Requirement already satisfied: PyYAML<6,>=5.4.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (5.4.1)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Filament-sensor-simplified==0.3.1) (1.2.0)
[... cut ...]
Building wheel for Filament-sensor-simplified (setup.py): finished with status 'done'
Created wheel for Filament-sensor-simplified: filename=Filament_sensor_simplified-0.3.1-py3-none-any.whl size=15614 sha256=12a492f3efb36c86696fb33c4185f585aa545eaa7ed950320acb4eb1aa083b5f
Stored in directory: /tmp/pip-ephem-wheel-cache-atisk3c3/wheels/a1/93/7b/976bb5c34ff5f4876d353afad4ed2f48f180a8e559157d7f73
Successfully built Filament-sensor-simplified
Installing collected packages: Filament-sensor-simplified
Successfully installed Filament-sensor-simplified-0.3.1
Fertig!

IHateNameStealers, 2022-05-23:
Doing a 3 day print at this moment (plugin was really needed this time).
When done, I will reinstall and try again.
Thank you for the info.