LuckyX182/Filament_sensor_simplified #59: Forcing Safe Mode Start Due to Plugin Error

therealalexsays, 2022-07-21:
Every other time I reboot, or turn system on, plugin will trigger an error and cause it to restart in safe mode. Strangely enough, rebooting will cause it to open fine and load all plugins the next time. Getting the following error:

2022-07-20 20:44:22,125 - py.warnings - WARNING - /home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentsensorsimplified/__init__.py:198: RuntimeWarning: No channels have been set up yet - nothing to clean up!  Try cleaning up at the end of your program instead!
  GPIO.cleanup()

After disabling, safe mode starts remedy themselves. Using 0.3.3. Had been using 0.3.1 before, and was occurring with that also. Love the plugin, sad it's not working right now.

LuckyX182, 2022-07-25:
I don't think a single warning could cause the Octoprint to boot into safe mode - tried it myself, I get the same warning but plugin works fine. Can you post your whole octoprint.log file?

DavidTolan, 2023-01-07:
I am seeing the same problem, I also have the enclosure plugin installed which uses GPIO to turn on a lamp. I have a spare raspberry pi, I'll try to set up just octopi and this plugin, and set up both plugins to see if I can reliably reproduce the problem