LuckyX182/Filament_sensor_simplified #66: Can't save settings

derekantrican, 2022-12-30:
Just installed this plugin. Connected the sensor. Restarted Octoprint. Set Pin number to "11", Switch type to "Triggered when closed" and clicked "Save". Upon opening the settings back up, none of the settings were saved.

octoprint.log shows the following error:

```
2022-12-30 05:22:39,141 - octoprint.plugins.filamentsensorsimplified - INFO - Saving settings for Filament Sensor Simplified
2022-12-30 05:22:39,150 - octoprint.server.api.settings - ERROR - Could not save settings for plugin Filament Sensor Simplified (0.3.3)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/settings.py", line 1139, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_filamentsensorsimplified/__init__.py", line 284, in on_settings_save
    usage = GPIO.gpio_function(pin_to_save)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
```

derekantrican, 2022-12-30:
Ok, upon changing the "Board mode" to "GPIO/BCM mode" and saving again, it seems to have saved. But now it seems to be stuck on that setting:

[image]

What does this mean?

Also, seems like something with the default value needs to be fixed with that dropdown if the saving isn't working

loroff, 2023-01-25:
bonjours moi j avais le meme probleme j ai desinstallé le plug in Exclude Region et apres sa fonctionné pourquoi ? je ne sais pas